jschuh/klipper-macros #190: Z-offset not resetting to 0 after SAVE_CONFIG

HitLuca, 2023-10-30:
I'll start by mentioning that I am currently not using the multiple bed surfaces feature of this project, so I may be missing some key details that would make this issue non existent in the first place.

When I have a z-offset change on my mainsail page and decide to save it, I just click SAVE_CONFIG. The printer.cfg file properly updates, but upon page reload I still see the same z offset. On a default mainsail installation the offset should be reset to 0 since it's integrated in the probe z offset, but when using this set of macros it doesn't. Why does this happen?

HitLuca, 2023-11-07:
Could you explain why this is intended behavior? I'm pretty sure that it has to do with bed surfaces but I'm not grasping the inner working

jschuh, 2023-11-10:
It's because there's always a default bed sheet (even if you don't configure it). Originally that wasn't the case, but I kept accidentally breaking that code path because I didn't use it, hence why it now just always uses the bed sheet code.

tastyratz, 2023-11-23:
I also had noticed this and I wasn't sure if I was doing something wrong. It seems odd as a default behavior, does that mean it's going to keep stacking on config saves if I don't clear it?

HitLuca, 2023-11-23:
@tastyratz yes, whenever you save a config with an updated z offset it will be restored to its value after firmware restart, but the endstop/probe positions will be updated as well. Clear it if you are not using the bed surfaces functionalities