jschuh/klipper-macros #199: Checking for extruder instead of heater in temperature menus

KRTac, 2023-11-09:
Hi

When there's a chamber section in the menu temps section, the gcode of preheat all checks for an extruder in the printer object, instead oh heater_generic chamber. [Example](https://github.com/jschuh/klipper-macros/blob/01c0d8aa06b39576501faaf9c905fae37d8aaef0/optional/lcd_menus.cfg#L407)

Stops preheat from triggering if there's no chamber heater present, like on my printer.

jschuh, 2023-11-10:
Oh yeah, that's a bug, but I can't imagine how it could stop the preheat from triggering. The only effect it should have is to allow a chamber setting a chamber temps from the menu when no chamber heater is actually configured.

KRTac, 2023-11-10:
I use chamber temp settings and don't have an active heater setup, only a `temperature_sensor chamber` that is referenced in a custom macro. That's also why the chamber temp doesn't get forwarded to KM's PRINT_START macros, because there is no heater to actually activate.

jschuh, 2023-11-29:
Should now be fixed in 088b4d0.

closed 2023-11-29 by jschuh