jschuh/klipper-macros #229: [BUG] Too Much Retract on PRINT_END for Microswiss Hotends

SSPW, 2024-01-29:
I am getting too much anti ooze rectract at the end of prints and clogging.
Looks like there are 3 lines with retraction for a total of 7mm (1 + 1 + 5) mm, which is way too much for a microswiss hotends.
I cant edit the macro...

Could this be made a parameter? If not, how do i unlock and edit the macro and make it stick?

[gcode_macro _print_end_inner]
variable_cancelled: False
gcode:
  SET_GCODE_VARIABLE MACRO=_print_end_inner VARIABLE=cancelled VALUE="{False}"
  {% set km = printer["gcode_macro _km_globals"] %}
  {% set toolhead = printer.toolhead %}
  {% set origin = printer.gcode_move.homing_origin%}
  {% set max_x = km.print_max[0] - origin.x %}
  {% set max_y = km.print_max[1] - origin.y %}
  {% set max_z = toolhead.axis_maximum.z - origin.z %}

  {% if not cancelled %}
    _KM_PRINT_STATUS ACTION=CHANGE STATUS=completing
  {% endif %}

  {% if printer.extruder.can_extrude %}
    # Wipe if we're not cancelling a paused print.
    {% if not printer.pause_resume.is_paused and not cancelled and
          toolhead.homed_axes|lower == "xyz" %}
      {% set x_safe = (max_x - toolhead.position.x, 2.0)|min %}
      {% set y_safe = (max_y - toolhead.position.y, 2.0)|min %}
      {% set z_safe = (max_z - toolhead.position.z, 2.0)|min %}
      G91
      G0 Z{z_safe} **E-1.0** F{km.travel_speed_z * 2} ; move nozzle up
      G0 X{x_safe} Y{y_safe} **E-1.0** F{km.travel_speed_xy} ; remove stringing
      # Remove the offset now that we're done.
      _KM_APPLY_PRINT_OFFSET RESET=1
    {% endif %}

    # Small retract to prevent ooze
    G92 E0
    **G1 E-5.0 F3600**
    M400
  {% endif %}

gretel, 2024-01-29:
making the retraction settings configurable in a generic way be cool. thank you

Heratiki, 2024-02-21:
Are you using a MicroSwiss Direct Drive setup as well as the MicroSwiss hotend?

jschuh, 2024-03-31:
I had this derived from `load_priming_length` locally, and just pushed that to dev in 5292fbf. That stated, feature requests belong in the [discussion section](https://github.com/jschuh/klipper-macros/discussions/categories/ideas).

closed 2024-03-31 by jschuh