jschuh/klipper-macros #231: [BUG] Option 'relative_reference_index' is not valid in section 'bed_mesh'

M3HNGRY, 2024-02-15:
Updated Klipper to v0.12.0-111-g4f00f219 running klipper-macros v0.8.8-0-g14840a4f. 

[image]

Gamithon, 2024-02-28:
To fix this after updating klipper:
Remove the depreciated "relative_reference_index" from you [bed_mesh] gcode (inside printer.cfg)
I also had to remove the "relative_reference_index: #" line as well
It is recommended to use "zero_reference_position" 

I didn't see any functionality differences after making the above changes. Is there any major differences between the two commands?

gretel, 2024-03-19:
is this project maintained?

jschuh, 2024-03-31:
As noted above, this is unrelated to the macros; you're getting it because Klipper removed `relative_reference_index`.

closed 2024-03-31 by jschuh