jschuh/klipper-macros #235: Klipper update breaks beeper

HitLuca, 2024-03-13:
Due to a recent klipper update, it seems that beepers now need to be defined as `pwm_cycle_time` elements ([forum link](https://klipper.discourse.group/t/after-latest-update-to-v0-12-0-98-m300-beeper-doesnt-work-properly/13523/6)).
While this fixes the issue with beepers not working and just emitting an audible _click_, the `M300` macro doesn't work anymore as it's looking for an `output_pin beeper` element in the printer config. Updating the macro check to also look for the new element would solve this issue.

Here is my quick fix to the the `M300` macro working again

```yaml
[gcode_macro m300]
description: Emits and audible beep.
  Usage: M300 [P<duration>] [S<frequency>]
gcode:
  {% set settings = printer.configfile.settings %}
  {% if "output_pin beeper" in printer or "pwm_cycle_time beeper" in printer %}
    {% set P = (params.P|default(100)|int, 0)|max %}
    {% set S = (params.S|default(1000)|int, 1)|max %}

    {% if "output_pin beeper" in printer %}
      SET_PIN PIN=beeper VALUE={% if settings["output_pin beeper"].pwm %}{
        settings["output_pin beeper"].scale|default(1.0) * 0.5
      } CYCLE_TIME={ 1.0 / S }{% else %}1{% endif %}
    {% else %}
      SET_PIN PIN=beeper VALUE={settings["pwm_cycle_time beeper"].scale|default(1.0) * 0.5} CYCLE_TIME={ 1.0 / S }
    {% endif %}
    
    G4 P{P}
    SET_PIN PIN=beeper VALUE=0
  {% else %}
    {action_respond_info(
       "M300 is disabled. To enable create an [output_pin beeper] or [pwm_cycle_time beeper] (new klipper versions) config.")}
  {% endif %}

```

jschuh, 2024-03-31:
Should be fixed in f90a118 on dev.

jschuh, 2024-04-01:
Fixed on main.

closed 2024-04-01 by jschuh