jschuh/klipper-macros #242: [BUG] _km_check_and_set_print_bounds macro fails if `MESH_MIN` or `MESH_MAX` have negative values

HitLuca, 2024-03-30:
I have a gcode file that has this generated `PRINT_START` code: 

```
PRINT_START EXTRUDER=215 BED=60 MESH_MIN=-0.232331,0.805592 MESH_MAX=119.524,120.17 LAYERS=150 CHAMBER=0
```

When trying to print it I get this error

```
Error evaluating 'gcode_macro _km_check_and_set_print_bounds:gcode': TypeError: not all arguments converted during string formatting
```

Since this issue never happened to me I'm fairly certain it's the `-0.232331` value in my `MESH_MIN` parameter which for some reason doesn't properly cast to float. I'm not sure what could be done, probably catching the error and displaying it in a more meaningful way (`your mesh bounds are negative and cannot proceed`) or bypass it would be nice.

jschuh, 2024-03-31:
Commit f90a118 on dev should fix it.

jschuh, 2024-04-01:
Fixed on main.

closed 2024-04-01 by jschuh