jschuh/klipper-macros #254: Feature: set temperature while meshing

pheidrias, 2024-07-09:
Hi,

it would be great to be able to set a nozzle-temperature to be held while probing the bed.

Atm, my filament starts oozing while meshing and I usually have the temperature fixed at 150°C.

I didn't find an option/config to do so - could you please implement it?

Thanks,
pheidrias

stablestud, 2024-07-27:
What about the following variable in `globals.cfg`:
```
# If non-zero the extruder will stabilize at this temp before probing the bed.
variable_start_extruder_probing_temp: 0
```

I believe it does exactly that

stablestud, 2024-07-27:
For additional confusion:
There is also the variable which is set by default:
```
# Extruder scale factor during pre-warmup in PRINT_START.
variable_start_extruder_preheat_scale: 0.5
```
I would expect it to have the same effect as `variable_start_extruder_probing_temp`, but it only sets the temp in stage PREHEAT and is replaced by final temperature in stage PROBING, unlike the `variable_start_extruder_probing_temp` which is replaced by the final temp only after probing has finished.

I would prefer the scale method also is only replaced at the end of probing not before.

Consider this:
```
variable_extruder_probing_temp = 120
variable_extruder_preheat_scale = 0.5
```
- Printing PLA:
  - Preheat temp: 100C (200C * 0.5)
  - Probe temp: 120C
  - This works good
- Printing PC:
  - Preheat temp: 150C (300C * 0.5)
  - Probe temp: 120C
  - Nozzle takes time to cool down, wasting time

Removing probing temp will lead to oozing, like described in the OP's issue.
Therefore I would propose to make `preheat_scale` behave exactly the same as `probing_temp`.

@jschuh what do you think?

stablestud, 2024-07-27:
Or a better solution would be to add variables for PREHEAT and PROBING:
```
variable_start_extruder_preheat_temp
variable_start_extruder_preheat_scale
variable_start_extruder_probing_temp
variable_start_extruder_probing_scale
```
This wouldn't break existing configs an adds further control to temps while probing.

closed 2025-01-14 by jschuh