jschuh/klipper-macros #255: [BUG] MAKE_SURFACE_MESH fails with "Move out of range"

stablestud, 2024-07-26:
**Describe the bug**

Seems like Klipper interprets negative zero `-0.000` less than `0.000` and therefore an out-of-bound 
error is thrown.

[image]

**Relevant macro output** (read from bottom to top)
```
Move out of range: -0.000 37.500 5.000 [0.000]
probe at 14.369,37.500 is z=0.902625
[...]
34  | (0.0, 55.0)     | (55.0, 65.0)
33  | (-0.0, 37.5)    | (55.0, 47.5)
32  | (14.4, 37.5)    | (69.4, 47.5)
[...]
bed_mesh: generated points
Index |  Tool Adjusted  |   Probe
Generating new points...
Stabilizing bed at 70C
```

**Relevant Klipper config**
```
[probe]
samples: 3
x_offset: 55
y_offset: 10

[bed_mesh]
mesh_min: 55, 30
mesh_max: 285, 310
probe_count: 9, 9
algorithm: bicubic
```

stablestud, 2024-07-27:
I think a solution would be to negate -0.000, so it results to 0.000

jschuh, 2025-01-14:
There's almost certainly a very small numerical precision error accumulating in either the macro's calculation or in Klipper's (it's just being truncated in the printed output). Unfortunately, this kind of happens due to the inherent imprecision of floating point values. The easiest solution is to just pad the limits of the mesh bounds a bit (e.g. by adding a small value for `variable_probe_mesh_padding` in your config).

closed 2025-01-14 by jschuh